neurobionics/opensourceleg #115: Position Control Tutorial

zachbons, 2023-09-21:
## 🐛 Bug Report

When I run the code in the Position Control Tutorial, nothing happens to the motor (it doesn't change position).

## 🔬 How To Reproduce

Steps to reproduce the behavior:

1. Type out the code in the tutorial word for word and run on hardware.

### Environment

* OS: 
Windows SSH'd into an RPi (Linux)
* Python version, get it with:
3.9.2

## 📈 Expected behavior

I expect the motor to move to the desired setpoint position

## 📎 Additional context

This is solved by placing osl.update() in the line before osl.knee.set_motor_position(). Also, the set_point description says that it's in 'motor ticks' but I believe it's actually the designated position units.

imsenthur, 2023-09-21:
@jderosia

jderosia, 2023-09-25:
Thank you for pointing this out. This pull request should fix that issue. #116

imsenthur, 2023-09-25:
Merged #116 with `master` [68196600c69b2a4b167c1b23bfb1252f67e346be].

closed 2023-09-25 by imsenthur