neurobionics/opensourceleg #131: Improved commenting at the top of parent files containing instructions for use.

tkevinbest, 2023-10-17:
## 🚀 Feature Request

It would be nice to add a comment block at the top of each file (such as `state_machine.py`) that describes how the modules should be used. For example, right now the `State` class is the first thing someone sees when they open the file which should never really be used on its own (unless I'm missing something). Basically I think we need something in each file that says how to use the components contained within that file in the intended way.

tkevinbest, 2023-10-17:
Similarly, it'd be helpful to know that `logger.py` is referenced by the OSL class and you don't have to instantiate it manually.

imsenthur, 2024-05-06:
Fixed with Jace's PR!

closed 2024-05-06 by imsenthur