neurobionics/opensourceleg #138: Actpack Offline Mode Failing for Thermal Limit

tkevinbest, 2023-10-18:
## 🐛 Bug Report

When running in offline mode, the loop fails after 3 iterations because the spoofed case temperature reading violates the maximum. The culprit is the new `read()` method that just increments all data by 15. 

## 🔬 How To Reproduce

Instantiate an OSL object in offline mode and run the loop a few times. You'll get this error: ` WARNING: [KNEE] Thermal limit 80 reached. Stopping motor.`

### Code sample: Culprit:

```
    def read(self):
        self._data.batt_volt += 15
        self._data.batt_curr += 15
        self._data.mot_volt += 15
        self._data.mot_cur += 15
        self._data.mot_ang += 15
        self._data.ank_ang += 15
        self._data.mot_vel += 15
        self._data.mot_acc += 15
        self._data.ank_vel += 15
        self._data.temperature += 15
        self._data.genvar_0 += 15
        self._data.genvar_1 += 15
        self._data.genvar_2 += 15
        self._data.genvar_3 += 15
        self._data.genvar_4 += 15
        self._data.genvar_5 += 15
        self._data.accelx += 15
        self._data.accely += 15
        self._data.accelz += 15
        self._data.gyrox += 15
        self._data.gyroy += 15
        self._data.gyroz += 15
```

### Environment

* OS: [WSL2]

imsenthur, 2023-10-18:
Edited those values to make more sense. Please checkout the dev branch [5e8c5c841c27ab3da91637e719a4fc7dffe607db].

closed 2023-10-18 by imsenthur

imsenthur, 2023-10-18:
Lowered the noise magnitude [0f76124d4cb557b93d0f77213e3a08244207732f].