neurobionics/opensourceleg #204: Accessing IMU data from the actuator

anushkarathi, 2024-04-19:
Allowing users to access the IMU data from the actuator 

IMU connections are having some issues and if I could access the IMU data from the actuator itself it would reduce the number of IMUs I would require and can give me a good estimate of the angle that I am looking for on the body (specifically I am using this in an ankle exo project and could estimate shank angle from the IMU on-board the actuator)! 

Alternatives would be to use another IMU- may be more cost-effective and efficient to simply use the IMU onboard the actuator

tkevinbest, 2024-04-22:
Further, we should also support providing the user with the raw actuator data in case they want to get something we don't provide natively. Perhaps `knee.raw_data` or similar?

imsenthur, 2024-05-06:
We already support IMU values streamed from a Dephy Actpack, please checkout these lines: https://github.com/neurobionics/opensourceleg/blob/5452e232109124c948b339fa39960c2c7c61dfc8/opensourceleg/hardware/actuators.py#L879-L943

We do relay all the data from the actpack as separate attributes but defining a custom packet that just returns the raw data from the actpack is a great idea but I think generalizing actuator classes is gonna change things a bit as not all motor drivers support this functionality so we'll have to instantiate a dummy IMU class and relay the data. Generalization will fix this!

closed 2024-05-06 by imsenthur

senthurayyappan, 2025-06-04:
@all-contributors add @anushkarathi as a contributor for bug report