neurobionics/opensourceleg #205: Callback function not working?

QiYue-day, 2024-04-28:
## 🐛 Bug Report

## 🔬 How To Reproduce

Steps to reproduce the behavior:

1. fsm_walking_controller.py

### Code sample
    def estance_to_lstance(osl: OpenSourceLeg) -> bool:
        assert osl.loadcell is not None
        return bool(

            osl.loadcell.fz < LOAD_LSTANCE
            and osl.ankle.output_position > ANKLE_THETA_ESTANCE_TO_LSTANCE
        )
    fsm.add_transition(
        source=early_stance,
        destination=late_stance,
        event=foot_flat,
        callback=estance_to_lstance,
    ）

### Environment

* OS: [e.g. Linux / Windows / macOS]
ubuntu22.04
* Python version, get it with:

```bash
python --version
```
Python 3.10.12
### Screenshots

## 📈 Expected behavior

## 📎 Additional context

When testing the above file, I found that my callback functions don't seem to be working, which resulted in the state machine still executing transitions when certain conditions weren't met. I want to verify if there's an issue with the callback functions.

tkevinbest, 2024-04-29:
Hi @QiYue-day,

Thanks for filing an issue! Were you able to resolve your problem? If not, we're happy to help. If so, can you please write up a quick blurb on this thread in case someone in the future has a similar issue? 

Thanks!

QiYue-day, 2024-04-30:
I haven't resolved my issue yet. However, upon closer examination of the callback function usage, I couldn't find any obvious issues with it. Even setting the callback function to False didn't prevent the state transition. I suspect it might be due to a hardware environment issue. If so, please let me know. If not, I hope you can help me resolve this problem.
Thanks！

QiYue-day, 2024-05-06:
yes

tkevinbest, 2024-05-06:
Have you confirmed that your OSL's sensor values are reasonable? For example, validated the load cell?

tkevinbest, 2024-05-06:
Hi. We found a small bug regarding units in the example script and have pushed an update. However, this bug made the switching criteria never met. Once fixed, the script operates as expected on our OSL. Printing out sensor values during the transitions would help me debug further.

QiYue-day, 2024-05-07:
Yes, I solved the problem, thank you for your help!

tkevinbest, 2024-05-07:
Can you share what the solution was, in case someone else runs into the same thing?

QiYue-day, 2024-05-08:
Sure, I modified the callback function invocation to allow it to be called even when there's no hardware, thus achieving my goal. Of course, I don't recommend using it this way because it might lead to some issues when connecting hardware later on.

closed 2024-05-08 by QiYue-day