neurobionics/opensourceleg #241: Get updated SoftRealtimeLoop

tkevinbest, 2024-06-18:
## 🚀 Feature Request

Get latest soft realtime loop from NLmiddleware. Test it in OSL applications.

senthurayyappan, 2025-03-06:
@tkevinbest Can you point me to the latest soft realtime loop? Is this already done in [#241-get-updated-softrealtimeloop](https://github.com/neurobionics/opensourceleg/tree/241-get-updated-softrealtimeloop)?

tkevinbest, 2025-03-07:
Shoot, I think so but we've found some issues in the latest softrealtime loop: https://github.com/UM-LoCoLab/NeuroLocoMiddleware/issues/29

I haven't looked into this further, but I could look closer next week.

tkevinbest, 2025-03-13:
We will want to merge from main NeurolocoMiddleware. There still might be something funky with increasing DT errors on non-pi platforms, but we should be okay for this application. It would be worth some testing to double check.

closed 2025-03-25 by senthurayyappan