neurobionics/opensourceleg #243: Dephy always logging

Katharine-Walters, 2024-06-21:
## 🐛 Bug Report

FlexSEA DebugLog is always being created regardless of the value of the dephy_log optional argument in the add_joint function. 

## 🔬 How To Reproduce

self.dev.add_joint(name="ankle", gear_ratio=9, port=self.actuator_port, dephy_log=False)

Setting dephy_log = True and dephy_log = False results in the same behavior

### Environment

* OS: [e.g. Linux / Windows / macOS]
* Python version, get it with:

```bash
python --version
```

### Screenshots

[image]

## 📈 Expected behavior

dephy_log = False does not create DebugLog file

Katharine-Walters, 2024-06-25:
Upon further investigation, I have discovered a few things: 
-The "dephy_log" bool is passed to the "debug" optional argument in the FlexSEA constructor. Is it unclear from the documentation what this controls, and I was unable to discern a change in behavior/logging when the bool was set to True or False.
-I was able to disable the dephy logging by setting the debug_level=6, which disables the logging in the FlexSEA constructor. 

I recommend implementing the following behavior: 
-Set default debug_level=4, which is the default logging level in the FlexSEA constructor. 
-When dephy_log = False, set debug_level=6, which will disable Dephy logging entirely.

closed 2024-10-09 by imsenthur

senthurayyappan, 2025-06-04:
@all-contributors add @Katharine-Walters as a contributor for bug report