neurobionics/opensourceleg #245: Joint Velocity property is wrong

tkevinbest, 2024-06-24:
## 🐛 Bug Report

The joint velocity property has the wrong units and the wrong direction for using Dephy's joint encoder. Dephy reports the units in deg/sec*10 and we're assuming it is in counts/sec. The direction is also incorrect. 

In actuators.py: 
```
    @property
    def joint_velocity(self) -> float:
        """Measured velocity from the joint encoder in rad/s."""
        if self._data is not None:
            return float(self._data.ank_vel * RAD_PER_COUNT)
        else:
            return 0.0
```

should be replaced with:
```
    @property
    def joint_velocity(self) -> float:
        """Measured velocity from the joint encoder in rad/s."""
        if self._data is not None:
            return float(self._data.ank_vel / 10.0 * DEG_2_RAD)
        else:
            return 0.0
```

tkevinbest, 2024-06-24:
So Dephy reports that it is *10, but in my testing it looks like this isn't true... perhaps something to do with the 9:1 gear ratio? Qualitatively, it looks like `return float(self._data.ank_vel * DEG_2_RAD)` would be correct.

imsenthur, 2024-06-26:
Added `return float(self._data.ank_vel * RAD_PER_DEG) * self.joint_direction` to the joint_velocity property. I don't why there is gear ratio mixed with the joint encoder velocity, we should test this out.

tkevinbest, 2024-06-26:
I'm overall confused still. This is what we get from your latest push. I would expect the velocity signal to be more centered in the noise of the position diff:
[image]

I don't know why their docs say it is multiplied by 10 when it clearly isn't... but it also doesn't look quite right.

tkevinbest, 2024-06-27:
Ok I'm dumb. My loop rate was 200 Hz, not 300. Just using deg/s looks right. 
[image]

commit 67e03d92 (2024-06-27): Merge pull request #246 from neurobionics/245-joint-velocity-property-is-wrong

Fixes #245: Joint velocity property
closed 2024-06-27 by tkevinbest