neurobionics/opensourceleg #376: Installation from source fails in ubuntu distro due to to adafruit-circuitpython-bno055 dependency

senthurayyappan, 2025-04-04:
**Describe the bug**
It looks like the dep fails on Ubuntu due to the adafruit-circuitpython-bno055 5.4.15 CircuitPython library for the BNO055 9-DOF absolute orientation sensor, the **sysv-ipc** is the one that fails

├── adafruit-blinka *
│   ├── adafruit-circuitpython-typing * 
│   │   ├── adafruit-blinka * (circular dependency aborted here)
│   │   ├── adafruit-circuitpython-busdevice * 
│   │   │   ├── adafruit-blinka >=7.0.0 (circular dependency aborted here)
│   │   │   └── adafruit-circuitpython-typing * (circular dependency aborted here)
│   │   ├── adafruit-circuitpython-requests * 
│   │   │   ├── adafruit-blinka * (circular dependency aborted here)
│   │   │   └── adafruit-circuitpython-connectionmanager * 
│   │   │       └── adafruit-blinka * (circular dependency aborted here)
│   │   └── typing-extensions >=4.0,<5.0 
│   ├── adafruit-platformdetect >=3.70.1 
│   ├── adafruit-pureio >=1.1.7 
│   ├── binho-host-adapter >=0.1.6 
│   │   └── pyserial * 
│   ├── pyftdi >=0.40.0 
│   │   ├── pyserial >=3.0 (circular dependency aborted here)
│   │   └── pyusb >=1.0.0,<1.2.0 || >1.2.0 
│   └── sysv-ipc >=1.1.0

senthurayyappan, 2025-04-04:
**adafruit-circuitpython-lis3dh 5.2.3 CircuitPython library for LIS3DH accelerometer.** also has **sysv-ipc** dep

├── adafruit-blinka *
│   ├── adafruit-circuitpython-typing * 
│   │   ├── adafruit-blinka * (circular dependency aborted here)
│   │   ├── adafruit-circuitpython-busdevice * 
│   │   │   ├── adafruit-blinka >=7.0.0 (circular dependency aborted here)
│   │   │   └── adafruit-circuitpython-typing * (circular dependency aborted here)
│   │   ├── adafruit-circuitpython-requests * 
│   │   │   ├── adafruit-blinka * (circular dependency aborted here)
│   │   │   └── adafruit-circuitpython-connectionmanager * 
│   │   │       └── adafruit-blinka * (circular dependency aborted here)
│   │   └── typing-extensions >=4.0,<5.0 
│   ├── adafruit-platformdetect >=3.70.1 
│   ├── adafruit-pureio >=1.1.7 
│   ├── binho-host-adapter >=0.1.6 
│   │   └── pyserial * 
│   ├── pyftdi >=0.40.0 
│   │   ├── pyserial >=3.0 (circular dependency aborted here)
│   │   └── pyusb >=1.0.0,<1.2.0 || >1.2.0 
│   └── sysv-ipc >=1.1.0 
└── adafruit-circuitpython-busdevice *
    ├── adafruit-blinka >=7.0.0 
    │   ├── adafruit-circuitpython-typing * 
    │   │   ├── adafruit-blinka * (circular dependency aborted here)
    │   │   ├── adafruit-circuitpython-busdevice * (circular dependency aborted here)
    │   │   ├── adafruit-circuitpython-requests * 
    │   │   │   ├── adafruit-blinka * (circular dependency aborted here)
    │   │   │   └── adafruit-circuitpython-connectionmanager * 
    │   │   │       └── adafruit-blinka * (circular dependency aborted here)
    │   │   └── typing-extensions >=4.0,<5.0 
    │   ├── adafruit-platformdetect >=3.70.1 
    │   ├── adafruit-pureio >=1.1.7 
    │   ├── binho-host-adapter >=0.1.6 
    │   │   └── pyserial * 
    │   ├── pyftdi >=0.40.0 
    │   │   ├── pyserial >=3.0 (circular dependency aborted here)
    │   │   └── pyusb >=1.0.0,<1.2.0 || >1.2.0 
    │   └── sysv-ipc >=1.1.0 
    └── adafruit-circuitpython-typing * (circular dependency aborted here)

unshrawal, 2025-05-27:
@senthurayyappan I tried replicating this issue, but I wasn't able to get any dependency issues. I tried building without the precompiled wheels with `pip install --no-binary :all: opensourceleg`, but I wasn't able to reproduce the issue. I also was able to setup the dev environment without any issues using Poetry. Could you clarify how you're installing the package from source?

senthurayyappan, 2025-06-03:
Hi @unshrawal, thanks for picking up this issue! I was installing the library from source via poetry on a Raspberry Pi 5 running ubuntu 24.04 aarch64 and here is the output I get:

```
Creating virtualenv opensourceleg in /home/senthura/opensourceleg/.venv
Installing dependencies from lock file

Package operations: 96 installs, 0 updates, 0 removals

  - Installing six (1.16.0)
  - Installing markupsafe (2.1.5)
  - Installing mergedeep (1.3.4)
  - Installing platformdirs (4.3.6)
  - Installing python-dateutil (2.9.0.post0)
  - Installing pyyaml (6.0.2)
  - Installing adafruit-circuitpython-connectionmanager (3.1.2)
  - Installing click (8.1.7)
  - Installing ghp-import (2.1.0)
  - Installing jinja2 (3.1.4)
  - Installing markdown (3.7)
  - Installing mkdocs-get-deps (0.2.0)
  - Installing packaging (24.2)
  - Installing pathspec (0.12.1)
  - Installing pyyaml-env-tag (0.1)
  - Installing watchdog (4.0.2)
  - Installing adafruit-circuitpython-busdevice (5.2.10)
  - Installing adafruit-circuitpython-requests (4.1.8)
  - Installing jmespath (1.0.1)
  - Installing mkdocs (1.6.1)
  - Installing pyserial (3.5)
  - Installing pyusb (1.2.1)
  - Installing typing-extensions (4.12.2)
  - Installing urllib3 (2.2.3)
  - Installing adafruit-circuitpython-typing (1.11.1)
  - Installing adafruit-platformdetect (3.75.0)
  - Installing adafruit-pureio (1.1.11)
  - Installing binho-host-adapter (0.1.6)
  - Installing botocore (1.35.59)
  - Installing colorama (0.4.6)
  - Installing mkdocs-autorefs (1.2.0)
  - Installing msgpack (1.0.8)
  - Installing numpy (1.26.4)
  - Installing pyftdi (0.55.4)
  - Installing pymdown-extensions (10.12)
  - Installing sysv-ipc (1.1.0): Failed

PEP517 build of a dependency failed

Backend subprocess exited when trying to invoke build_wheel

    | Command '['/tmp/tmp3_5uvvxf/.venv/bin/python', '/home/senthura/.local/share/pypoetry/venv/lib/python3.12/site-packages/pyproject_hooks/_in_process/_in_process.py', 'build_wheel', '/tmp/tmpqdjljyvk']' returned non-zero exit status 1.
    | 
    | /tmp/tmp3_5uvvxf/.venv/lib/python3.12/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
    | !!
    | 
    |         ********************************************************************************
    |         Please consider removing the following classifiers in favor of a SPDX license expression:
    | 
    |         License :: OSI Approved :: BSD License
    | 
    |         See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
    |         ********************************************************************************
    | 
    | !!
    |   self._finalize_license_expression()
    | ******************************************************************************
    | * Setup can't determine the value of PAGE_SIZE on your system, so it will
    | * default to 4096 which may not be correct.
    | * 
    | * Please report this message and your operating system info to the package
    | * maintainer listed in the README file.
    | ******************************************************************************
    | running bdist_wheel
    | running build
    | running build_ext
    | building 'sysv_ipc' extension
    | creating build/temp.linux-aarch64-cpython-312
    | aarch64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -I/tmp/tmp3_5uvvxf/.venv/include -I/usr/include/python3.12 -c common.c -o build/temp.linux-aarch64-cpython-312/common.o
    | common.c:2:10: fatal error: Python.h: No such file or directory
    |     2 | #include "Python.h"
    |       |          ^~~~~~~~~~
    | compilation terminated.
    | error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1

Note: This error originates from the build backend, and is likely not a problem with poetry but one of the following issues with sysv-ipc (1.1.0)

  - not supporting PEP 517 builds
  - not specifying PEP 517 build requirements correctly
  - the build requirements are incompatible with your operating system or Python version
  - the build requirements are missing system dependencies (eg: compilers, libraries, headers).

You can verify this by running pip wheel --no-cache-dir --use-pep517 "sysv-ipc (==1.1.0)".

  - Installing wrapt (1.16.0)
  - Installing zipp (3.20.2)
```

and it looks like sysv-ipc package has trouble installing on ubuntu-24.04-aarch64. Here is some more information on the system:

```
Linux raspberrypi 6.8.0-2019-raspi-realtime #20-Ubuntu SMP PREEMPT_RT Thu Feb 27 14:16:18 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux
```

senthurayyappan, 2025-06-03:
I tried working around this issue by activating the poetry virtual env and pip installing the sysv-ipc package:

```
senthura@raspberrypi:~/opensourceleg$ eval $(poetry env activate)
(opensourceleg-py3.12) senthura@raspberrypi:~/opensourceleg$ pip install sysv-ipc
Collecting sysv-ipc
  Using cached sysv_ipc-1.1.0.tar.gz (99 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: sysv-ipc
  Building wheel for sysv-ipc (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for sysv-ipc (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [31 lines of output]
      /tmp/pip-build-env-4gbqqksa/overlay/lib/python3.12/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
      !!
      
              ********************************************************************************
              Please consider removing the following classifiers in favor of a SPDX license expression:
      
              License :: OSI Approved :: BSD License
      
              See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
              ********************************************************************************
      
      !!
        self._finalize_license_expression()
      ******************************************************************************
      * Setup can't determine the value of PAGE_SIZE on your system, so it will
      * default to 4096 which may not be correct.
      *
      * Please report this message and your operating system info to the package
      * maintainer listed in the README file.
      ******************************************************************************
      running bdist_wheel
      running build
      running build_ext
      building 'sysv_ipc' extension
      creating build/temp.linux-aarch64-cpython-312
      aarch64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -I/home/senthura/opensourceleg/.venv/include -I/usr/include/python3.12 -c common.c -o build/temp.linux-aarch64-cpython-312/common.o
      common.c:2:10: fatal error: Python.h: No such file or directory
          2 | #include "Python.h"
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for sysv-ipc
Failed to build sysv-ipc

[notice] A new release of pip is available: 25.0.1 -> 25.1.1
[notice] To update, run: pip install --upgrade pip
ERROR: Failed to build installable wheels for some pyproject.toml based projects (sysv-ipc)
```

closed 2025-08-25 by senthurayyappan