neurobionics/opensourceleg #440: No installation setup includes spidev, which is needed for NBLoadcellDAQ

tkevinbest, 2025-06-23:
**Describe the bug**
None of the following installations install `spidev`, but it is a dependency required for the `NBLoadcellDAQ`:

[image]

**To Reproduce**
Steps to reproduce the behavior:

1. Start with a fresh image and install all features. `spidev` isn't included. 

**Expected behavior**
Some installation flag should include this. It would also be nice if there was an `[all]` option to just do all at once.

tkevinbest, 2025-07-18:
@unshrawal is this fixed in a branch somewhere? I didn't see a PR, but it looks like you fixed it.

unshrawal, 2025-07-19:
Hi @tkevinbest, I’ve committed a fix for this issue that works with Poetry as the package manager, but it will break once we migrate to uv. I plan to update it accordingly once the uv migration PR is merged.

unshrawal, 2025-07-21:
No, that commit is still on a branch on my fork. I haven't created a pull request yet since this is still going to be an issue once the uv issue is merged.

unshrawal, 2025-08-14:
This issue seems to be resolved with PR #438 Migrate to UV.